OttoDIY/OttoDIYPython #16: optimization (HEAP/RAM usage)

sfranzyshen, 2020-09-23:
### Things are moving fast ... thanks to everyone pitching in 👍 

our target platform is the esp32, but we should support the esp8266 as well. While the esp32 and esp32s2 have adequate RAM (especially units that have PSRAM) the esp8266 is a bit more limited. We should consider optimizing our development for both speed as well as efficiency. Additionally, we should consider releasing our project as a complete firmware binary release with our code preinstalled ready to go.

### Internal Constant Values
One optimization is the use of const values for integers or numbers that don't ever change.  There's a special const function added to MicroPython that allows you to tell the interpreter and other tools that a value will never change.  When tools know the value is constant they can optimize its usage, like inlining the value where it's used instead of creating a variable that takes precious memory.  As your saw in the driver code page it makes sense to set register and other fixed values as consts.

### Compile Modules into Frozen Bytecode
When a developer creates a Micro Python application they are creating scripts that are executed by the Python interpreter at run-time. Processing the script at run-time will not only use space on the heap but can also cause the heap to become fragmented. Developers can take modules within their application code and cross compile it into bytecode that is stored in flash with the kernel code. Creating the bytecode will cause the modules code to execute from flash rather than executing from the heap. The result is that less heap space is used and there is decreased heap fragmentation. Constant values and strings can also be precompiled into frozen bytecode which will prevent them from taking up unnecessary RAM.

### Custom Built Firmware With Additional Modules 
build a module inside the firmware; this option optimizes the module execution and minimizes the size of the same on the device; the module is frozen into the firmware.

### References:
https://docs.micropython.org/en/latest/reference/constrained.html#execution-phase
https://docs.micropython.org/en/latest/library/micropython.html#module-micropython
https://www.beningo.com/5-tips-for-optimizing-the-heap-in-micropython/
https://learn.adafruit.com/porting-an-arduino-library-to-circuitpython-vl6180x-distance-sensor?view=all
https://www.microdev.it/wp/en/2018/06/25/micropython-micropython-compiling-for-esp8266/
https://www.microdev.it/wp/en/2018/07/04/micropython-micropython-compiling-for-esp8266-with-additional-modules/

sfranzyshen, 2020-09-23:
there isn't a simple method to work with list (arrays) ... but there is a solution

https://github.com/micropython/micropython/issues/4067

mishafarms, 2020-09-27:
Even though notes are floats, When we apply them to tone, we force them to be ints. So we might as well store them as ints
Put in a comment on the real float values just to let everyone know.

sfranzyshen, 2020-09-27:
> Even though notes are floats, When we apply them to tone, we force them to be ints. 

very interesting I never looked ...

> So we might as well store them as ints Put in a comment on the real float values just to let everyone know.

agreed ...